looooo/freecad.gears #140: Failed to install freecad.gears

Shalmeneser, 2023-12-25:
Problem with Addon Manager : https://forum.freecad.org/viewtopic.php?t=83791

> 
> Failed to install freecad.gears
> Git returned a non-zero exit status: 128
> Called with: /usr/bin/git clone --recurse-submodules https://github.com/looooo/freecad.gears /home/shalmeneser/.local/share/FreeCAD/Mod/freecad.gears
> 
> Returned stderr:
> Clonage dans '/home/shalmeneser/.local/share/FreeCAD/Mod/freecad.gears'...
> error: 1078 octets de corps sont encore attendus
> fetch-pack: unexpected disconnect while reading sideband packet
> fatal: fin de fichier prématurée
> fatal: fetch-pack : sortie d'index de pack invalide

Shalmeneser, 2023-12-26:
No problem one day later.

looooo, 2024-01-01:
can we close this?

closed 2024-01-20 by looooo