looooo/freecad.gears #145: Using thornton cutters and measurments

alexkoel1958, 2024-01-24:
I'm drawing clocks and has the problem that the measurments of the gears aren't completely ok. I'm using the Thornton tables and formulas. 
For instance:
Gear with 100 teeth and a module of 0,8.
The outside or tip diameter should be 82,2 mm (teeth+2,76)*module. In FreeCAD it ends up at 81,7 mm!
The root diameter should be 76,8 mm (teeth-4)*module. In FreeCAD it ends up at 77,99 mm!
The teeth width should be 1,256 mm (1,57*module). It is 1,23 mm! This widt is also on at to large root diameter. 

Is it possible to make the cycloidial gear part to be custimized.

I know that a lot of great guys are building FreeCAD and all off the Addons. But this extra feature should be great.
If interested I can supply the thornton formulas. They are also depending on the module and the number off teeth with pignions (teeth <= 16).

looooo, 2024-01-27:
You can adjust the teeth height with the head parameter and the root can be adjusted with the clearence parameter.
The formulas used can be found here: https://github.com/looooo/freecad.gears/blob/master/pygears/cycloid_tooth.py#L39
and here: https://github.com/looooo/freecad.gears/blob/master/pygears/cycloid_tooth.py#L40

alexkoel1958, 2024-01-28:
[image]
Still working from Z=100 and M=0,8
See the formula's of Thornton. With all due respect. If I calculate the Outside diameter (Da) from the formula in the source with a head off zero I get 82,4.. If I make a techdraw, the outside is over the teeth is 81,7. So somewhere it's going wrong. Not that it's a big issue. I can manualy change the measurements. But it would be great if I understand if I'm doing something wrong or the code :-).