looooo/freecad.gears #146: Error when saving file with gear

viktak, 2024-01-27:
I just installed this WB for the latest FreeCAD (0.21.2) as I need to add a timing gear to my project. The gear appear fine, I can change its properties, all is good, but when I save the project, I get the following error in the 'Report view':
```
17:53:56  PropertyPythonObject::toString(): failed for <class 'freecad.gears.timinggear.TimingGear'>
17:53:56  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.21\bin\lib\json\__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "C:\Program Files\FreeCAD 0.21\bin\lib\json\encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "C:\Program Files\FreeCAD 0.21\bin\lib\json\encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "C:\Program Files\FreeCAD 0.21\bin\lib\json\encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type FeaturePython is not JSON serializable
17:53:56  PropertyPythonObject::toString(): failed for <class 'freecad.gears.basegear.ViewProviderGear'>
17:53:56  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.21\bin\lib\json\__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "C:\Program Files\FreeCAD 0.21\bin\lib\json\encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "C:\Program Files\FreeCAD 0.21\bin\lib\json\encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "C:\Program Files\FreeCAD 0.21\bin\lib\json\encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type ViewProvider is not JSON serializable
```
While trying to figure out what causes this I noticed that any gear type causes the same problem.

Is it possible that the current version of the WB is not compatible with the latest version of FreeCAD? Or something is wrong in my environment? All other WBs work fine on my computer. My environment is a 64 bit Windows 11.

Thank you in advance for any pointers!

looooo, 2024-01-27:
The addon-mqnager doesnt use the latest version. Try installing freecad.gears manually.

viktak, 2024-01-28:
Thanks for the quick reply!
I removed it from the addon-manager, then I installed it manually, but it doesn't even show up in the workbenches dropdown list. I tried to install it using both pip and pip3, but no luck, still FreeCAD doesn't see it...

looooo, 2024-01-28:
It depends on how you installed freecad.

viktak, 2024-01-28:
Simply downloaded the Windows installer and installed it...

looooo, 2024-01-28:
Ok then you need to install freecad.gears with the same python that is also shipped with freecad. Or you simple copy to the right place ..../lib/sitepackages

viktak, 2024-01-28:
In FreeCAD's About box I can see python 3.8.10. So when I tried to install it with pip 3, it should have worked, right?

This is the output of the install:
```
PS D:\> pip3 install https://github.com/looooo/FCGear/archive/master.tar.gz
Collecting https://github.com/looooo/FCGear/archive/master.tar.gz
  Using cached https://github.com/looooo/FCGear/archive/master.tar.gz
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: numpy in c:\users\viktor\appdata\local\programs\python\python310\lib\site-packages (from freecad.gears==1.2.0) (1.24.3)
Requirement already satisfied: scipy in c:\users\viktor\appdata\local\programs\python\python310\lib\site-packages (from freecad.gears==1.2.0) (1.12.0)
Requirement already satisfied: sympy in c:\users\viktor\appdata\local\programs\python\python310\lib\site-packages (from freecad.gears==1.2.0) (1.12)
Requirement already satisfied: mpmath>=0.19 in c:\users\viktor\appdata\local\programs\python\python310\lib\site-packages (from sympy->freecad.gears==1.2.0) (1.3.0)
Building wheels for collected packages: freecad.gears
  Building wheel for freecad.gears (pyproject.toml) ... done
  Created wheel for freecad.gears: filename=freecad.gears-1.2.0-py3-none-any.whl size=208747 sha256=c0921881412553a0a303f70e13d5007e6d17317185567f0ad3d007b0cb1373fb
  Stored in directory: C:\Users\Viktor\AppData\Local\Temp\pip-ephem-wheel-cache-a3gx_ffc\wheels\b4\4b\5c\f2259e212822d70e6570319d345308f5ffd03d697867b4967d
Successfully built freecad.gears
Installing collected packages: freecad.gears
Successfully installed freecad.gears-1.2.0
```

I can even find it in the site-packages folder:
[image]

It still wouldn't show up within FreeCAD. Is there anything else to do in FreeCAD to make it work? (I have restarted it)

looooo, 2024-01-28:
it seems like you installed it to python 3.10. If freecad uses python 3.8.10 it will not see the installed freecad.gears.

viktak, 2024-01-28:
I see, thanks for explaining this to me!

arampak, 2024-02-03:
To follow up on this. When saving the file with a generated gear, I had an error when the workbench was installed using the built-in addon manager. Here is the error message in the console:

```
13:03:28  PropertyPythonObject::toString(): failed for <class 'freecad.gears.basegear.ViewProviderGear'>
13:03:28  Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type ViewProviderDocumentObject is not JSON serializable
```
I have just tried installing the gear workbench in FreeCAD on MacOS using pip. For that, I have cd'd into /Applications/FreeCAD.app/Contents/Resources/bin/ and run:
./python -m pip install https://github.com/looooo/FCGear/archive/master.tar.gz 

Everything was installed without problem, however, the same error remains.

./python --version gives prints: "Python 3.10.13"

The file is actually saved, and the generated gear is there after re-opening, but probably with some information loss.

looooo, 2024-02-11:
@chennes do you know why some people have problems install the latest version of this workbench with the addonmanager?

chennes, 2024-02-11:
I just installed it with the Addon Manager and it worked fine (though I don't know how to test that the WB itself is working, I just hit install, restarted, then launched it).

viktak, 2024-02-12:
@chennes See the OP: just create a timing gear and save it. Then you'll see (or not :) ) the issue.

turbubestia, 2024-03-02:
Same problem here.

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Spanish/Chile (es_CL)
Installed mods: 
  * Assembly3 0.12.2
  * freecad.gears 1.2.0
```

The next exception is triggered when saving the project. Also, it is triggered for every gear, not just once.

```
20:32:26  PropertyPythonObject::toString(): failed for <class 'freecad.gears.basegear.ViewProviderGear'>
20:32:26  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.21\bin\lib\json\__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "C:\Program Files\FreeCAD 0.21\bin\lib\json\encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "C:\Program Files\FreeCAD 0.21\bin\lib\json\encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "C:\Program Files\FreeCAD 0.21\bin\lib\json\encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type ViewProvider is not JSON serializable
```

av2k, 2024-03-24:
Can confirm with Python 3.11:

```
Sketcher constraint number 1 is malformed!
The Sketch has malformed constraints!
Recompute failed!
PropertyPythonObject::toString(): failed for <class 'freecad.gears.features.ViewProviderGear'>
Traceback (most recent call last):
  File "__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "encoder.py", line 200, in encode
    chunks = self.iterencode(o, _one_shot=True)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "encoder.py", line 258, in iterencode
    return _iterencode(o, 0)
           ^^^^^^^^^^^^^^^^^
  File "encoder.py", line 180, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type ViewProvider is not JSON serializable
```

stepheneb, 2024-05-15:
Another report FreeCAD 0.21.2.33771 using Python 3.10.13  on macOS 14.4.1

Steps:
1. open freecad
2. make new document
3. open gear workbench
4. make worm gear
5. save document

FreeCAD About
```
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * ThreadProfile 1.89.0
  * freecad.gears 1.2.0
  * A2plus 0.4.64a
  * sheetmetal 0.4.14
  * Beltrami 1.0.8-alpha
  * Curves 0.6.36
  * Assembly4 0.50.12
  * Assembly3 0.12.2
  ```

Error: `<class 'TypeError'>: Object of type ViewProviderDocumentObject is not JSON serializable`

Listedd errors

```
Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type ViewProviderDocumentObject is not JSON serializable

PropertyPythonObject::toString(): failed for <class 'freecad.gears.basegear.ViewProviderGear'>

Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/json/encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type FeaturePython is not JSON serializable

PropertyPythonObject::toString(): failed for <class 'freecad.gears.wormgear.WormGear'>
```

Python installed on system

```
% python3 --version 
Python 3.12.3
% which python3
/opt/homebrew/bin/python3
```

[image]

stepheneb, 2024-05-15:
Same errors saving when installing latest manually as described in [#146#issuecomment-2111358476](https://github.com/looooo/freecad.gears/issues/146#issuecomment-2111358476)

1. Use addon-manager to uninstall freecad.gears
2. Install freecad.gears manually

```
% cd ~/Library/Application\ Support/FreeCAD/Mod
% git clone git@github.com:looooo/freecad.gears.git 
% cd freecad.gears 
% git log -1

commit 9715c068e4fb98f9e60cba8eea97980a73e60fae (HEAD -> master, origin/master, origin/HEAD)
Author: catalintucureanu <catalintucureanu@gmail.com>
Date:   Thu May 9 22:17:18 2024 +0300

    Update timinggear_t.py
    
    fix for issue #164
```

SGStino, 2024-07-22:
I'm still seeing ` PropertyPythonObject::toString(): failed for <class 'freecad.gears.basegear.ViewProviderGear'>` 

with a git clone into the %appdata%/FreeCAD/Mod folder.

python on the system was 3.11, freecad uses 3.8.

Issue happens both with version https://github.com/looooo/freecad.gears/commit/6197bfdc16a927be15212aa8eed5b7fe9378e45d as with master.

To me, it looks like it's trying to serialize something it shouldn't be serializing.

features.py/ViewProviderGear or InvoluteGear looks like they are not meant to be serialized with json, but i don't know enough about freecad's internals to know what it should do instead, or how it should be able to convert it into json.

SGStino, 2024-07-24:
I downloaded FreeCAD about a week or two ago from the website, installed gears through the addon manager, and i had the issue. I then uninstalled gears, and installed via git clone into the appdata folder, and it was still there.

When i'm back at that device i can look at what the version number of FreeCAD was, but i wonder if that's something they could have broken, unless they are now saving a lot more objects than they used to do in the release I used?

looooo, 2024-07-24:
hmm strange.
I cannot reproduce with:
```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Austria (de_AT)
Installed mods: 
  * DynamicData 2.62.0
  * Plot 2022.4.17
  * freecad.gears 1.2.0
  * AirPlaneDesign 0.4.1
  * CurvedShapes 1.0.9
  * Curves 0.6.42
  * Assembly4 0.50.15
  * Assembly3 0.12.2
```

and 

```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Austria (de_AT)
Installed mods: 
  * DynamicData 2.62.0
  * Plot 2022.4.17
  * freecad.gears 1.2.0
  * AirPlaneDesign 0.4.1
  * CurvedShapes 1.0.9
  * Curves 0.6.42
  * Assembly4 0.50.15
  * Assembly3 0.12.2
```

is this platform specific?

SGStino, 2024-07-25:
The error would suggest not, it's json serialization of an object, which would never work on any platform?
but i'm on Windows, if you aren't, then it might?

I'll see if i can reproduce it on another machine this weekend.

SGStino, 2024-07-25:
I asked someone with a mac to try:
[image]

```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
```

So it doesn't seem to be platform dependant.
It still might be python >3.11 then. I'll have to figure out how to let freecad use another version.

Sol-Ell, 2024-11-10:
### Issue Summary
Error related to `ViewProviderGear` when encoding or decoding this class to JSON in FreeCAD. The error arises due to the current implementation’s version-dependent methods.

### System Details
- **OS**: Windows 11 (build irrelevant)
- **FreeCAD Version**: 0.21.2.33771 (Git)
- **Word size**: 64-bit
- **Python version**: 3.10.13 (embedded in FreeCAD, no separate Python installation required)
- **Installed Mods**: `freecad.gears` v1.3.0

### Example Error Output
```plaintext
[Time]  PropertyPythonObject::toString(): failed for <class 'freecad.gears.basegear.ViewProviderGear'>
[Time]  Traceback (most recent call last):
  File "C:\Users\[Username]\Desktop\FreeCAD\bin\lib\json\__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "C:\Users\[Username]\Desktop\FreeCAD\bin\lib\json\encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  ...
<class 'TypeError'>: Object of type ViewProviderDocumentObject is not JSON serializable
```

### Solution Steps
1. **Locate Addon Path**
   - Go to **Tools** -> **Addon Manager** in FreeCAD.
   - Find the **`freecad.gears`** workbench and note the **Installation Location**. (Example path: `C:\Users\[Username]\AppData\Roaming\FreeCAD\Mod\freecad.gears`).

2. **Find the Class Definition**
   - Navigate to the `freecad.gears` directory in your file explorer.
   - Locate the file containing `ViewProviderGear` (usually found in `[Path to addon]\freecad\gears\basegear.py`).

3. **Modify the Class Code**
   - Open the file and locate the following class definition:
   ```python
   class ViewProviderGear:
       """
       The base Viewprovider for the gears
       """

       # [Some other code here...]

       if sys.version_info[0] == 3 and sys.version_info[1] >= 11:
           def dumps(self):
               self._check_attr()
               return {"icon_fn": self.icon_fn}

           def loads(self, state):
               if state and "icon_fn" in state:
                   self.icon_fn = state["icon_fn"}
       else:
           def __getstate__(self):
               self._check_attr()
               return {"icon_fn": self.icon_fn}

           def __setstate__(self, state):
               if state and "icon_fn" in state:
                   self.icon_fn = state["icon_fn"}
   ```
   - Remove the version check condition (`if sys.version_info[0] == 3...`) and adjust the indentation:
   ```python
   class ViewProviderGear:
       """
       The base Viewprovider for the gears
       """

       # [Some other code here...]

       def dumps(self):
           self._check_attr()
           return {"icon_fn": self.icon_fn}

       def loads(self, state):
           if state and "icon_fn" in state:
               self.icon_fn = state["icon_fn"}

       def __getstate__(self):
           self._check_attr()
           return {"icon_fn": self.icon_fn}

       def __setstate__(self, state):
           if state and "icon_fn" in state:
               self.icon_fn = state["icon_fn"}
   ```
4. **Save the File and Restart FreeCAD**

### Explanation of Fix
The `if` condition in the original code limits the use of `dumps` and `loads` methods to Python 3.11+, while providing only `__getstate__` and `__setstate__` for earlier versions. Removing this condition makes both sets of methods available universally, resolving the JSON serialization issue.

looooo, 2024-11-10:
thanks, applied with https://github.com/looooo/freecad.gears/commit/4d87de3f0cc4ad75d5db617814b13695becc8330

Sol-Ell, 2024-11-11:
Thank you for your quick response. I noticed the error also occurs when creating and saving new gears, but it temporarily resolves after saving the project and reloading FreeCAD.

The following errors appear:

- **`<class 'TypeError'>: Object of type InvoluteTooth is not JSON serializable`** – occurs when saving a project with a newly created `Involute Gear` or `Internal Involute Gear`.
- **`<class 'TypeError'>: Object of type InvoluteRack is not JSON serializable`** – occurs when saving a project with a newly created `Involute Rack`.
- **`<class 'TypeError'>: Object of type CycloidTooth is not JSON serializable`** – occurs when saving a project with a newly created `Cycloid Gear`.
- **`<class 'TypeError'>: Object of type FeaturePython is not JSON serializable`** – occurs when saving a project with newly created gears such as `Cycloid Rack`, `Crown Gear`, `Worm Gear`, `Timing Gear T-shape`, `Timing Gear`, `Lantern Gear`, and `Hypo Cycloid Gear`.
- **`<class 'TypeError'>: Object of type BevelTooth is not JSON serializable`** – occurs when saving a project with a newly created `Bevel Gear`.
- **`<class 'TypeError'>: Object of type ViewProviderDocumentObject is not JSON serializable`** – occurs after combining two gears in the project (`Combine Two Gears`).

Additional details on the final error:
```plaintext
[Time]  PropertyPythonObject::toString(): failed for <class 'freecad.gears.connector.ViewProviderGearConnector'>
[Time]  Traceback (most recent call last):
  File "C:\Users\[Username]\Desktop\FreeCAD-0.21.2-Windows-x86_64\FreeCAD_0.21.2-2023-12-17-conda-Windows-x86_64-py310\bin\lib\json\__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
  File "C:\Users\[Username]\Desktop\FreeCAD-0.21.2-Windows-x86_64\FreeCAD_0.21.2-2023-12-17-conda-Windows-x86_64-py310\bin\lib\json\encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "C:\Users\[Username]\Desktop\FreeCAD-0.21.2-Windows-x86_64\FreeCAD_0.21.2-2023-12-17-conda-Windows-x86_64-py310\bin\lib\json\encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "C:\Users\[Username]\Desktop\FreeCAD-0.21.2-Windows-x86_64\FreeCAD_0.21.2-2023-12-17-conda-Windows-x86_64-py310\bin\lib\json\encoder.py", line 179, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type ViewProviderDocumentObject is not JSON serializable
```