looooo/freecad.gears #156: Ondsel Freecad crash with freecad.gears

turbubestia, 2024-03-05:
I tried to use freecad.gear on the latest Ondsel but it crashes with no message, except and `abnormal termination` in the log.

```
OS: Windows 11 build 22631
Word size of Ondsel: 64-bit
Version: 2024.1.0.35694 (Git)
Build type: Release
Branch: flavor
Hash: b070090c6873b3fcf4be9a459219d22edfc4e7cf
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Spanish/Chile (es_CL)
Installed mods: 
  * freecad.gears 1.2.0
  * Ondsel-Lens 2024.1.22.01
  * OpenDark 2023.12.17
```
 If I run the next code in the Readme.md

```python
import FreeCAD as App
import freecad.gears.commands
gear = freecad.gears.commands.CreateInvoluteGear.create()
gear.teeth = 20
gear.beta = 20
gear.height = 10
gear.double_helix = True
App.ActiveDocument.recompute()
Gui.SendMsgToActiveView("ViewFit")
```

it crashes in the `App.ActiveDocument.recompute()`

It would be great to have this plugins working in Ondsel Freecad.

[Ondsel.log](https://github.com/looooo/freecad.gears/files/14489849/Ondsel.log)

0phiTeS, 2024-03-29:
I face the exact same issue. I opened an issue on the ondsel repository. 
It works fine for me on vanilla freecad but not in ondsel. 
It also crashes in the App.ActiveDocument.recompute()  for me also.

looooo, 2024-06-19:
@turbubestia can we close this issue?

turbubestia, 2024-06-20:
I tested it today and it works with gears version 1.2.0 and Ondsel version 2024.2.1.

Thanks.

closed 2024-06-20 by turbubestia