looooo/freecad.gears #157: Selecting Involute Gear in V.22 Development Branch Causes Immediate Crash of Freecad

BrianAtDocumentedDesigns, 2024-03-05:
I installed the workbench for FreeCad. In the current release version for Windows 

([code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * 3D_Printing_Tools
  * A2plus 0.4.61
  * freecad.gears 1.2.0
  * Help 1.0.3
[/code]

 it works fine. If I use it from the V.22 development branch

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.36277 (Git)
Build type: Release
Branch: main
Hash: 9e1903d46112b3660bf10c6a4537d728101d560b
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * 3D_Printing_Tools
  * A2plus 0.4.61
  * freecad.gears 1.2.0
  * Help 1.0.3

And select the Involute Gear icon, FreeCad immediately crashes and shuts down. Involute Rack seems to work fine. Other options appear to also cause a crash.

maxwxyz, 2024-03-17:
can confirm, not possible to add involute gears.

hasecilu, 2024-05-02:
Is still crashing? I tested on lates weekly build (18 hours ago) and it works fine.

BrianAtDocumentedDesigns, 2024-05-02:
No, I checked it the other day and it seemed to work fine with the most
recent version of 22.

closed 2024-06-19 by looooo