looooo/freecad.gears #164: module 'scipy' has no attribute 'optimize'

aladdain, 2024-04-16:
This error happens whenever I try to add any gear type

```
03:08:12  Traceback (most recent call last):
  File "/home/salah/snap/freecad/common/Mod/freecad.gears/./freecad/gears/basegear.py", line 127, in execute
    gear_shape = self.generate_gear_shape(fp)
  File "/home/salah/snap/freecad/common/Mod/freecad.gears/./freecad/gears/lanterngear.py", line 101, in generate_gear_shape
    phi_min = sp.optimize.root(find_phi_min, (phi_max + r_r / r_0 * 4) / 5).x[
<class 'AttributeError'>: module 'scipy' has no attribute 'optimize'
03:08:12  Recompute failed!
```
[image]

Ubuntu 22.04
Python 3.10
FreeCad version: 0.21.2

mickdavis, 2024-04-30:
I'm getting the same error.  Is this workbench still supported?

hasecilu, 2024-05-02:
I tested all commands on latest FreeCAD weekly and all it's fine.

It could be that another WB included an old version of scipy when it doesn't had that attribute? Happened to me once.

catalintucureanu, 2024-05-09:
Also fails for me in Timing Gear with the same error.

Please add: 
`from scipy import optimize`
to the import section of timinggear_t.py to fix. No other issues apart from this.

Context: freecad-daily (0.22.0dev) under Linux mint (Ubuntu 22.04), python 3.10.12 and scipy 1.8.0

looooo, 2024-06-19:
should be fixed now.

closed 2024-06-19 by looooo