looooo/freecad.gears #168: Different method for entering parameters for Lantern Gear possible?

azalanono, 2024-06-17:
Hi, looo.

I wanted to design a sprocket for a bicycle chain. 

It was really complicated to enter right data. Finally I gave up and bought the chain first, to be able to test generated sprockets made by my 3D printer before laser cutting them from steel. So I am not at the beginning of trial and error until the sprocket really fits.

This rises a question:
A chain sprocket is usually defined by the pitch (distance of joints), diameter of rollers (on joint bolts), and number of teeth.

Would it be a huge work to alternatively allowing to enter above  values and generating the sprocket that way?

If I need free play between chain and sprocket, this should be a fourth parameter, making teeth narrower in tangential direction only.

What do you think?

looooo, 2024-06-19:
Not sure if it is possible to have two ways to specify the parameters, but it should be not a problem to define lantern-gears by a different set of parameters (if this is a more common use-case).

azalanono, 2024-06-19:
I needed 4 tries to find the right module for a 1/2" roller chain. It is "4.06". 
Calculating the module was unclear.

Is it possible to have two entry fields, where one automatically updates the other one on change (and vice versa)?

If yes, this would be an excellent way.

If I enter a chain-segment length, the module calculated and updated from it.   If the module is edited, the chain-segment length is calculated and updated from module.