looooo/freecad.gears #178: Changing the number of Teeth in a InvoluteGear doesn´t work

Raphy2907, 2024-10-09:
I am using FreeCad 1.0.0RC2 at the moment, but the problem also excists in my copy of version 0.21.1.
When i change the num_teeth calue from 15 to any other number, I get the following error message:

16:10:20  pyException: Traceback (most recent call last):
  File "C:\Users\Praktikum\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 145, in execute
    gear_shape = self.generate_gear_shape(fp)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Praktikum\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\involutegear.py", line 338, in generate_gear_shape
    profile = rotate_tooth(tooth, obj.num_teeth)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Praktikum\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 295, in rotate_tooth
    return part.Wire(flat_shape)
           ^^^^^^^^^^^^^^^^^^^^^
<class 'Part.OCCError'>: BRep_API: command not done
16:10:20  InvoluteGear: BRep_API: command not done

Can´t make much sense of that. Perhabs anyone else has expierenced a simular issue.

looooo, 2024-10-09:
should be fixed with: https://github.com/looooo/freecad.gears/commit/13b2a9c4a5ce63c5f8d79bc75db91cb89de706d7

please report back if it is working for you

motoz, 2024-10-10:
I noticed the same problem yesterday, now it works fine. Thanks!

closed 2024-10-10 by looooo