looooo/freecad.gears #179: Gear WB, Involute gear works only with 15 teeth

KariKKT, 2024-10-11:
Hi,  I am new to Freecad,  Trying to learn abt gears with workbench version 1.3.0.    Trying to change the number of teeth from 15 fails and get always this message "InternalInvoluteGear: BRep_Api: command not done".    All other gear types I tried seem to work fine.

Windows 11, Freecad 0.21.2 and 1.0.0RC1

Any advise, thanks

looooo, 2024-10-11:
This was recently fixed. Try to update the workbench

guillaumechauvat, 2024-10-15:
Hi, I just installed the gear workbench from the addon manager but I still have this issue. Is the version in the addon manager not the latest? Should I use pip instead?

guillaumechauvat, 2024-10-15:
It works by adding
`fp.gear.num_teeth = fp.num_teeth`
in internalinvolutegear.py at line 278.

Roy-043, 2024-10-21:
Another user has reported this issue and I can reproduce:
https://github.com/FreeCAD/FreeCAD/issues/17365

The fix proposed by @guillaumechauvat solves the problem.

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38977 (Git)
Build type: Release
Branch: main
Hash: facd8ca6e0303cf96297a7a7398185770c1a1e0b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * freecad.gears 1.3.0
```

luzpaz, 2024-10-21:
@looooo :point_up_2:

looooo, 2024-10-21:
should be fixed with the recent commits.

closed 2024-12-11 by looooo