looooo/freecad.gears #185: Wrong version number

skintigh, 2024-11-05:
After using bevelgear.teeth for months, yesterday it stopped working and I had to use .num_teeth.

This was odd as the version, 1.0.0, hadn't changed according to the addon manager and it said this was the up to date version. Yet the install date had changed to 11/3.

If I view FreeCAD's help->about info, the version is 1.3.0 not 1.0.0

I refreshed the cache in addon manager, and the version number updated, the date remained the same.

Not sure if this is a bug in FCGear, FC, or just my computer.

[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.15
  * DynamicData 2.68.0
  * freecad.gears 1.3.0
[/code]

luzpaz, 2024-12-09:
JFY @kintigh  
An addon's version/date info is sourced from the `package.xml` file  
https://github.com/looooo/freecad.gears/blob/26e8703270ac5518bbf867f9b06fc9e1889f15c7/package.xml#L5-L6

I'm not sure what the issue is. Perhaps it's related to you running FreeCAD 0.21.2  
Can you recreate with 1.0 [^1]?  

[^1]: https://blog.freecad.org/2024/11/19/freecad-version-1-0-released/

luzpaz, 2025-01-07:
@skintigh :point_up:

skintigh, 2025-01-07:
I have a lot of installed versions (4? 5?) on several different machines but that was the only time that happened. May have just been a fluke.

luzpaz, 2025-01-07:
Maybe better to close this issue ITC ?

closed 2025-01-07 by looooo