looooo/freecad.gears #194: model breaking after switching to freecad 1.0

jelle284, 2024-12-16:
I have a model which i created with freecad 0.21 and an older version of fc.gears (can't remember which one i had, as this problem occured after updating to the latest version).

I then opened it in freecad 1.0 and updated all my workbenches (including this one)

Upon recomputing the model i got an error on all my gears:
> <Exception> ObjectIdentifier.cpp(1534): Property 'teeth' not found in 'teeth'
in property binding 'gear_small#InvoluteGear008.dw'

I then notice that the property had changed name to num_teeth but the expression for dw remained
> teeth*module

After manually updating the expression, i get lots of new errors which i don't understand

> 20:51:14  pyException: Traceback (most recent call last):
  File "C:\Users\Jelle\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\basegear.py", line 142, in execute
    gear_shape = self.generate_gear_shape(fp)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Jelle\AppData\Roaming\FreeCAD\Mod\freecad.gears\.\freecad\gears\involutegear.py", line 329, in generate_gear_shape
    obj.gear.axle_hole = obj.axle_hole
                         ^^^^^^^^^^^^^
<class 'AttributeError'>: 'FeaturePython' object has no attribute 'axle_hole'

I tried uninstalling the workbench and installing again but no luck

looooo, 2024-12-17:
I guess we need to add some backward compatibility functions. Can you post the file, so we can work on this?

jelle284, 2024-12-17:
[gear_small.zip](https://github.com/user-attachments/files/18170550/gear_small.zip)
Here are the files to reproduce.

looooo, 2024-12-18:
Do you also have the file where you didn't manually updated from teeth -> num_teeth (the origin file which worked with an older version of freecad.gears? The version you posted have already gears which are not processed by the gear-workbench anymore.

I am sorry for this inconviniece. We made the change from teeth to num_teeth because of consistency, but i didn't expect it to make that much trouble. 

There are several issues:
- the property teeth was changed to num_teeth. But this should work anyways by some backward-compatibility function.
- the property dw was previously calculated by an expression. But this is not working for shifted gears, thats why I changed it to be calculated by a python function.
- the hole property was added (actually I wasn't a big fan of this, but I merged the commit, as I thought it's a advancement some people will use)

looooo, 2024-12-18:
if you pull the latest master (update the freecad.gears addon) the hole-properties shouldn't be a problem anymore. But I guess you will still run into the teeth->num_teeth issue.

@wwmayer do you have an idea to replace a obj in the tree by an updated obj. Mainly all the properties stay the same, only a read-only property needs to be updated as it is now not computed by an expression anymore.

wwmayer, 2024-12-19:
> do you have an idea how to replace an obj in the tree by an updated obj.

You don't have to replace objects in the tree. Instead the Python feature class has to implement the additional method `onDocumentRestored(self, obj)` where you adjust objects from old projects to the current status, i.e. add additional properties, remove deprecated properties, ...

Here some example code:

This is the initial implementation of MyFeature.
```
class MyFeature:
    def __init__(self, obj):
        obj.Proxy = self
        obj.addProperty("App::PropertyInteger", "teeth")
```
Here you use it in a document and save it to a project file
```
doc = App.newDocument()
obj = doc.addObject("App::FeaturePython", "MyFeature")
fea = MyFeature(obj)
obj.teeth = 5
doc.saveAs("/tmp/test.FCStd")
```

Now you have decided to change the property name.
```
class MyFeature:
    def __init__(self, obj):
        obj.Proxy = self
        obj.addProperty("App::PropertyInteger", "num_teeth")
```

But if you load the old project file it still uses the old property
```
doc = App.openDocument("/tmp/test.FCStd")
doc.ActiveObject.teeth
doc.ActiveObject.num_teeth # error
```

So, add the method onDocumentRestored to handle this case
```
class MyFeature:
    def __init__(self, obj):
        obj.Proxy = self
        obj.addProperty("App::PropertyInteger", "num_teeth")

    def onDocumentRestored(self, obj):
        if hasattr(obj, "teeth"):
            num_teeth = getattr(obj, "teeth")
            obj.addProperty("App::PropertyInteger", "num_teeth")
            obj.num_teeth = num_teeth
            obj.removeProperty("teeth")
```

Load the old project file again
```
doc = App.openDocument("/tmp/test.FCStd")
doc.ActiveObject.num_teeth # Works
doc.ActiveObject.teeth # Not available any more
```

jelle284, 2024-12-19:
Here are the original files created with FreeCAD v0.21
[gear_small_v021.zip](https://github.com/user-attachments/files/18202756/gear_small_v021.zip)
Thanks for the help and quick replies btw

looooo, 2024-12-21:
thanks @wwmayer this is exactly what I needed.
@jelle284 also with the file you posted it is for me impossible process the gears. Best would be to have only one Involutegear, that was made with the old version of freecad.gears.