looooo/freecad.gears #195: Planet gear not fitting

ChuckInGame, 2024-12-20:
Hi,

I have an issue with a planet gear.

you can see here, that in FreeCad, the gears teeth will not fit in the outer ring:

[image]

I tribble checked my calculation and i think, i did every thing right.

I thought, lets print it:

[image]

yeah, everything is fine in the real world. The planet gear works as calculated, but in freecad, its still displayed not fitting well, like in the first screenshot.

the FCStd file is here:

[planetengetriebe2.zip](https://github.com/user-attachments/files/18204316/planetengetriebe2.zip)

I tried several versions of freecad and ended up in the weekly:

OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39646 (Git) Conda
Build type: Release
Branch: main
Hash: b25b83c1a58bf0ae95cbd502d0ccae27e8127cad
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/Germany (en_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * fasteners 0.5.33
  * freecad.gears 1.3.0

I tried this with Ubuntu 24.10, with the same result.

What did I wrong, is this a bug or is the problem in front of the monitor :)

Regards,

Chuck

dzid26, 2025-02-01:
Your design would work with 4 planets.  But not 3.
2*(sun+planet) needs to be divisible by number of planets. https://docs.google.com/spreadsheets/d/1kx-RGIGDojEfSCOoIUq_cxSQF2gSpotjH5--TAVFono

Gear workbench doesn't have planet gear calculation https://github.com/looooo/freecad.gears/issues/163.
You should close this issue.

ChuckInGame, 2025-02-22:
Hi dzid25, 
thanks, that you answered to my issue.

T'm sorry, but i don't understand your google docs document.

Can you please explain me, which formular is wrong and what formular i have to use and where i can find a documentation of it ?

Regards,
Chuck

dzid26, 2025-02-22:
Your formulas are probably ok.  But you need also this constraint:
> 2*(sun teeth + planet teeth) needs to be divisible by number of planets.

You can use my document to quickly check your values. If it highlights red, that means it violates the constraint.

dzid26, 2025-03-07:
> if I understand correct, the teeth count and diameter of the planets are not depending on the count of planets.

If you want to distribute planets equally at a certain angle (120deg in case of 3 planets), the number of teeth cannot be random, else you will not align them properly with both sun and ring. That's exactly what happens in your first picture. 

So there are two constraints: 
- `ring_teeth = sun_teeth + 2*planet_teeth` - you probably already used that
- `2*(sun teeth + planet teeth) / CountOfPlanets` **needs to be an integer**.

If you don't care about having exact 120deg between the planets then you can ignore second constraint. But you still need to adjust the carrier..., so it's better to simply ensure the alignment.

ChuckInGame, 2025-03-07:
Thank you for the explanation; it makes much more sense to me now.

I will tackle this next week and try to fix it.

Once again, I really appreciate you taking the time.

Best regards,
Chuck

ChuckInGame, 2025-03-09:
I changed the parameters to ensure the constraint is an integer, and I got a wonderful gear!

[image]

Thanks a lot! I'll close this ticket now because it was my mistake.

closed 2025-03-09 by ChuckInGame