looooo/freecad.gears #196: Axle hole will not go all the way through depending on beta angle

dogunter, 2024-12-20:
I am using FreeCAD 1.0 with the geers workbench 1.3.0.

If I create a simple involute gear and keep all the settings at the defaul, then set axel_hole as true, I get the expected result. However, if I then set a beta angle value of 12º or higher, the bottom surface of the hole turns solid, the hole no longer cuts all the way through.

The beta value of 12º or more which demonstrates this varies depending on the overall size of the gear one needs, but it works to demonstrate the behavior I'm running into.

The workaround is to simply leave axle_hole set to false and create a pocket hole in the parts design workbench.

loughkb, 2024-12-20:
I can verify this behavior.  Beta angle results in the hole not going all the way through.   Came here to report it, but found this, so I thought I'd just add my observation as well.   Wonderful workbench by the way!  You guys have done an excellent job on it.  Very useful.  Thanks to all the devs.

looooo, 2024-12-21:
this is a result of a occt-issue. See this failing test:
https://github.com/looooo/freecad.gears/blob/master/tests/tests.py#L10

commit 284e2881 (2024-12-25): Extend axle and offset holes past the limits of the gear. Fixes #196
closed 2024-12-25 by looooo