looooo/freecad.gears #200: Not installable (1.1.0dev)

EmEiBee, 2025-02-01:
Hello all,

while I try to install Gears it crash with an error pointing to git.
I have no idea about git, so I also have no idea what's going wrong...

[image]

What can I also provide to fix that?

`OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.40041 (Git) Conda
Build type: Release
Branch: main
Hash: 64a4f6e3669279e5dbe9cb59d8e0123eaeb70833
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Logical/physical DPI: 96/110.189
Installed mods: 
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
`

looooo, 2025-02-01:
does this also happen with other addons?

EmEiBee, 2025-02-01:
... let me see ... (tic... toc ...)
Not this error. I have now try to install "TimingGear". That results into an error like...
`21:36:29  cannot import name '_iterative' from partially initialized module 'scipy.sparse.linalg.isolve' (most likely due to a circular import) (C:\Program Files\FreeCAD\bin\Lib\site-packages\scipy\sparse\linalg\isolve\__init__.py)`

Then I have installed "SheetMetall Workbench". That results into an error like...

`21:40:27  Command '['C:\\Program Files\\FreeCAD\\bin\\python.exe', '-m', 'pip', 'install', '--disable-pip-version-check', '--target', 'C:\\Users\\Administrator\\AppData\\Roaming\\FreeCAD\\Mod\\..\\AdditionalPythonPackages\\py311', 'networkx']' died with <Signals.CTRL_BREAK_EVENT: 1>.

EmEiBee, 2025-02-02:
I, clueless as I am, tried a bit..

.

1. Installed the current GIT version for Windows and then entered the path to it correctly in FC.

2. Tried to execute the clone command manually. The command is "C:\Program Files\Git\cmd>git.exe clone --recurse-submodules https://github.com/looooo/freecad.gears c:\users\Administrator\AppData\Roaming\FreeCAD\Mod\freecad.gears"
This works without errors, but only if the directory didn't exist before.

3. Deleted the directory manually and tried to install Gears from FC again. The error message remains, so nothing has changed.
4.

looooo, 2025-02-02:
Seems like there is something wrong with your installation.  If I would be you I would try pixi :) But this advise comes more from a package-manager view. I can give instructions if you want. But keep in mind that you will be an early user of pixi, so we might run into some issues. If you want to go this route, it's best to read a little bit about pixi: https://prefix.dev/. So just tell me if you want to go this road.

EmEiBee, 2025-02-02:
Oh... I'm already completely overwhelmed and have no idea how it all works; this will definitely going to go wrong! You overestimate my knowledge in this field...

In the meantime, I completely uninstalled GIT and Python from the system and made sure that there were no leftovers, so that FC has to use its own Git and Python.
After that, at least the installation went through without errors. So it's obviously a version problem if a newer GIT and/or Python version exists outside of FC.
But ultimately it didn't help either. Because when I start Gears, the issues mentioned above remain...

You write here "python > 3 (for python2 use branch py2)". Before that, I had Python 3.13.1 running externally on the system and referred to it in FC. However, I don't know which Python version FC itself has, so I can't judge whether that might be the cause.

Would it make sense to uninstall FC completely, install the official 1.0 and then copy the DEV's onto it?

looooo, 2025-02-02:
I really can't help with your setup. There is somehow a script issue. But I am not sure which scipy is used. I think we normally be ndle it with freecad, but not sure about this.
The problem often is when users want to add additional dependencies. But with the bundles this is not really supported. That's why I suggested to use a good package-manager like pixi, which allows to install freecad and additional dependencies like scipy and also ensures that the dependencies work together.

EmEiBee, 2025-02-02:
Yes, I understand you. But I'm afraid that I'll completely lose the thread with additional things.

Let me first try the whole uninstall, system clean, reinstall the official 1.0 and then install Gears. It'll probably take until tomorrow. But then at least I'll know whether the problem here is with the system (dependencies) or somewhere else...

EmEiBee, 2025-02-03:
Ok...
I had deleted the existing FC version including all user settings and then deleted all remaining folders and files, as well as existing entries in the registry.
I then installed the official 1.0 and installed all the plugins one by one.
Up until then everything was fine and everything worked.
Then I copied the current DEV over it and lo and behold: everything was still fine!

So it seems that somehow or somewhere there were still fragments of a pre-1.0 that prevented the plugins from working.

Thanks for the help...

(... if only the FC developers could finally introduce work paths at some point ...)