looooo/freecad.gears #201: [PartDesing] The Sketch loses the reference when the gear us changed.

bitcuits, 2025-02-25:
In the PartDesign when you add a sketch to the surface of the gear the sketch loses the reference to the gear because the surface of the gear changes names.

Example.
In the gear_witch.FCStd a sketch is added to the surface.  (This works fine).
The sketch refers to the surface named
IvotutionGear[Face92]

In the gear_with_sketch_lost_reference.FCStd here the gear dimensions have changed and the sketch loses the reference to the gear.
This error is displayed in freecad.
Sketch: No planar face in AttachEngine3D::calculateAttachedPlacement()!

When the sketch is deleted and a new sketch is drawn the drawing can be repaired. 
File gear_with_new_sketch.FCStd
It can be seen that the new sketch has now changed name to
IvotutionGear[Face122]

I believe that this could be fixed if the names of the gear surfaces are not changed.
Like IvolutionGearXXX.top and IvolutionGearXXX.bottom.

This fix could make the gears much more use-able in the PartDesign'er.

Thank for the good work.

looooo, 2025-02-25:
this is a topo-naming issue. I do not know if naming the bottom and top face manually is possible. If anyone knows, please let me know. Best practice is to not reference the faces directly.

bitcuits, 2025-02-25:
Ok yes I see.